yaacov/ArduinoModbusSlave #4: EEPROM Write Problem

Persp0lisy, 2017-09-26:
Dear Kobi Zamir,
Hi
Please accept my thanks for your very use full Library for ARDUINO modbus slave.
i successfully uploaded full.ino code in example of library and successfully : 
1.Read/Write digital pins.
2.Read Analog pins.
3.Read EEPROM registers.
and problem is here that i want to write in EEPROM using function 06(Write_Single_Register) because some Limitations.
Thanks for your support.

yaacov, 2017-09-26:
Ahh ... thanks

What I understand is that we are missing a `FC_WRITE_SINGLE_REGISTER` ?

here:
https://github.com/yaacov/ArduinoModbusSlave/blob/master/src/ModbusSlave.h#L24
https://github.com/yaacov/ArduinoModbusSlave/blob/master/src/ModbusSlave.cpp#L162

a. did I understand the issue correctly ?
b. can you make a pull request that add the missing option ?

yaacov, 2017-09-26:
I will try to look into adding this next weekend :-)
if you can do this yourself and do a [pull request](https://help.github.com/articles/about-pull-requests/) it will be nice.

Persp0lisy, 2017-09-26:
aaaaaaaaaaaaa yes that was just the problem :D 
Unfortunately i'm new in arduino and i dont think i can do it lonley.

Persp0lisy, 2017-09-26:
Thanks for your support very very much and GOOD LUCK

Persp0lisy, 2018-02-05:
Dear Kobi Zamir,
Hi
i would like to thank you a lot for your support,i checked your library now and my problem is solved.
thank you again and good luck
Yours sincerely

closed 2018-02-12 by Persp0lisy