yaacov/ArduinoModbusSlave #11: Exchanging strings

raffaeler, 2017-11-20:
This is not an issue but just a question.
What are, on your opinion, the best options to copy a string between the master and the slave?
Apparently, but I am not a modbus expert, it looks like the Read/Write File Record (0x14 / 0x15) are the best option in the specs.
Should I implement these services? What do you think?

yaacov, 2017-11-20:
> Should I implement these services?

a. That will be great !
b. If you just want to copy string ( and not implement new features, although that is fun too :-) ) you can use write and read multiple registers ( FC04, FC16 ) for specific address, and agree that this address represent a string.

e.g. 
for hello
send this registers: 68 65 6c 6c 6f ( + 00 pedding ) to address 0050:
`11 10 0050 0003 06 6865 6c6c 6f00 CRC0`

raffaeler, 2017-11-20:
The read/write multiple registers was already my fallback strategy :)
I was wondering whether you have any suggestions about implementing read/write file record... what is the gain on your opinion?

I have no device doing that, so the risk is writing a bad implementation that works just because I own both the master and the slave.
I am using QModMaster on a PC to test the slave implementation, but it does not support the read/write file record.

yaacov, 2017-11-20:
> I have no device doing that, so the risk is writing a bad implementation that works just because I own both the master and the slave.

I never worked with a device that used modbus for file transfer, usually if we needed to move files, we used other methods ( e.g. data card, tcp, other protocol ... ) so I can not offer an advice here, For small strings ( like a title in display ) we did use FC16.

raffaeler, 2017-11-20:
Agree, I'll see, thanks

closed 2017-11-20 by raffaeler

yaacov, 2017-11-20:
have fun, If you do decide to implement file read/write, It will be welcome even if it will a bad implementation :-) [ it will be better than no implementation at all ]