yaacov/ArduinoModbusSlave #20: Problem is the get INPUT registers

Apulanta, 2018-10-14:
Hi Yaacov Zamir

Thanks to the excellent Full modbus Arduino skelets you made.
I've added the NTC10 temperature calculation and PID control everything works as well and I can apply set values ​​in EEPROM memory ..

My only problem is the INPUT registers query
30001 = PID adjustment value (360-359 value)
30002 = NTC10 Temperature (22.6 - 22.7 C)
30003 = Blank
30004 = PWM calculation value (55-56%)
30005 = Blank

This would be my desired morbus register list.
But when I ask for one of the INPUT registers eg 30004 (57) I always get 30001 (360) value in the register.
I have tried to make the INPUT FC = 04 offset settings where I could get the correct value of it but I failed.

This writeRegisterToBuffer command confuses because I do not want to write the value directly to something I want in the registry. Do you need a new FUNKTION here?

You could look at my appendix pictures if they better tell you what it is all about.

Can you tell me how should I act to get the modbus register functions to work correctly?

I am a novice in these matters, and yet the desire is hard to understand programming and other technology packages .. Big thank you Modbus excellent program for you.

Sincerely, Rauno Finland
[image]
[image]
[image]
[full_TESTI.zip](https://github.com/yaacov/ArduinoModbusSlave/files/2476175/full_TESTI.zip)
[image]

yaacov, 2018-10-14:
@Apulanta Thank you for the issue, I do not think I will have time to look at it this week ... 

Can you look at code and find a solution ?
If you do please add a comment here, or better, open a pull request with the fix.

I will try to look at it, but do not have lot's of time lately :-(

Apulanta, 2018-10-14:
Thanks for your quick reply.
I would be very grateful if you could look at the zip package in the weeks or months, I'm also trying to find the solution to the problem yourself.
The bug is in the INPUT 04 registry part I have created ..

Handle reading input registers (FC = 04)
(for (int i = 0; i <length; i ++) {
        // enter the uint16_t value for the response buffer.
        // slave.writeRegisterToBuffer (i, analogRead (address + i));
         slave.writeRegisterToBuffer (0, analogRead (address = i));
         slave.writeRegisterToBuffer (1, address = TP); // Outdoor temperature
         slave.writeRegisterToBuffer (2, address = PMWoutputValue); // PWM% calculated
       // slave.writeRegisterToBuffer (3, (analogRead (address = 4)); // Modbus address 30002

Sincerely, Rauno to Finland

falahati, 2019-06-04:
Just for sake of google searches now since this issue is old;

@Apulanta, you need to write only the parts you need to the buffer. So what I suggest is to create an array containing all input values that you have, and then copy that array to the buffer based on the `address` and `length` of request.

In your example, you always return `360` as your first response no matter what the request asks for. Forget about 30000 and 30004; with your code, any address you request will result in the same response from the device.

closed 2019-09-03 by yaacov