yaacov/ArduinoModbusSlave #21: Problem is the get INPUT registers

Apulanta, 2018-10-15:
Hi Yaacov Zamir.

Sorry yet a new issue.
I made a simple example sketch from which you can easily see how the problem occurs, i did the "// EXAMPLES" commas tell the editing section.

In addition, it is custom edited by the full_Examples.ino sketch

Sincerely, Rauno Finland

[full_Examples.zip](https://github.com/yaacov/ArduinoModbusSlave/files/2480097/full_Examples.zip)

yaacov, 2018-10-15:
> Sorry yet a new issue.

a. No problem :-) thanks for the issues, that's how we make things better !
b. I will try to look at it, but it may take me some time, as I will have very little time, in the next few weeks.
c. f you can find the problem and fix it it will be very helpful.

Apulanta, 2018-10-17:
Ok .. Thank you for the message
I have tried two weeks to solve the problem, the problem is interesting and instructive.
I am a beginner of programming but I have a desire to get better ..

Regards, Rauno

yaacov, 2018-10-17:
Thanks for the update :+1:  if I find the time, I will look at it and update here.

Apulanta, 2018-10-30:
Hi Yaacov,

I got the problem solved, I made a new program for the irrigation timer you made and it seemed to me to be more functional and better.

It would be one more question while I'm still beginner in programming.
I would like to extend this irrigation program from the program
"register 104..109 - permanent user information"
I would like to make the area more extensive, for example 104 to 114. (+5 registers)

so are these changes sufficient.

#define NUM_REGISTERS 118 -> #define NUM_REGISTERS 123

// user persistent data
    for (i = 96; and <(96 + 7); i ++) {
        EEPROM.get (2 + i * 2, au16data [i + 7]);
change --->
// user persistent data
    for (i = 96; and <(96 + 12); i ++) {
        EEPROM.get (2 + i * 2, au16data [i + 12]);

and

   // user persistent data
    for (i = 96; and <(96 + 7); i ++) {
        EEPROM.put (2 + i * 2, au16data [i + 7]);
change --->
   // user persistent data
    for (i = 96; and <(96 + 12); i ++) {
        EEPROM.put (2 + i * 2, au16data [i + 12]);

In addition, AO0 read 110 ---> 115 changes of registers ..

I tried the change but there was something in the registry for EEPROM incompatibility and program crash ..

Sincerely, Rauno Finland

yaacov, 2018-10-30:
Hi, Thanks for the update !

a. Great.
b. I do not remember how the EEPROM [1] library work ... 
c. you can write small programs that only write and read from eeprom to see if you can understand what makes the program crash.

[1] https://www.arduino.cc/en/Reference/EEPROM

closed 2019-09-03 by yaacov