yaacov/ArduinoModbusSlave #23: add callback functions

zatalian, 2018-12-20:
Hi,

I would like to see the CB_READ_COILS function being split into CB_READ_COILS and CB_READ_DISCRETE_INPUTS.
The same for CB_READ_REGISTERS. This one should be split into CB_READ_INPUT_REGISTERS and CB_READ_HOLDING_REGISTERS.

I think this would make the programming of the slave code a lot more intuitive. Is there a reason why this is not done in the first place?

yaacov, 2018-12-20:
> I think this would make the programming of the slave code a lot more intuitive. Is there a reason why this is not done in the first place?

not that I can remember :-) you are welcome to make a pull request :+1:

yaacov, 2018-12-22:
Fixed in : https://github.com/yaacov/ArduinoModbusSlave/pull/24

closed 2018-12-22 by yaacov