yaacov/ArduinoModbusSlave #25: Change SLAVE_ID

g6pl, 2018-12-31:
Hello
How i can change SLAVE_ID in void loop or void setup?

g6pl, 2018-12-31:
Hello. You do not understand me ... :(  I would like to change the value of SLAVE_ID during program operation

g6pl, 2018-12-31:
This is not the fault of the library, but the lack of my knowledge of arduino ..

g6pl, 2018-12-31:
I would like the program to read the SLAVE_ID from the EEPROM memory and then run the Modbus slave (SLAVE_ID, CTRL_PIN); How can I run Modbus slave (SLAVE_ID, CTRL_PIN) in Void Setup or Void Loop?

yaacov, 2018-12-31:
No, the slave-id is set in the constructor [1] 

Once it is set, it can not be changed. If you add a new method to the `Modbus` class [2] that update the private `unitID` var [3], you will be able to change the unit id (that was set by SLAVE_ID in the constructor) while program is running ( e.g. inside loop or setup)

[1] https://github.com/yaacov/ArduinoModbusSlave/blob/master/examples/simple/simple.ino#L38
[2] https://github.com/yaacov/ArduinoModbusSlave/blob/master/src/ModbusSlave.h#L73
[3] https://github.com/yaacov/ArduinoModbusSlave/blob/master/src/ModbusSlave.h#L93

an example of such a method:
```
/**
 * Set Unit ID of modbus object.
 *
 * @param unitID the modbus slave id.
 */
void Modbus::setUnitID(uint8_t _unitID)
{
    // set modbus slave unit id
    unitID = _unitID;
}
```

yaacov, 2018-12-31:
Changing CTRL_PIN is more tricky :-) , the easiest way is to create a new modbus object with a new Serial stream [1] and a new ctrl pin.

Check carefully that you must get the ctrl_pin at runtime, it may get a little interesting if you have little experience with arduino.

[1] https://github.com/yaacov/ArduinoModbusSlave/blob/master/src/ModbusSlave.cpp#L39

g6pl, 2018-12-31:
In ModbusSlave.h i change:
`class Modbus {
public:
    Modbus(uint8_t unitID, int ctrlPin);
    Modbus(Stream &serial, uint8_t unitID, int ctrlPin);
    void begin(unsigned long boud);
    int poll();
    int readCoilFromBuffer(int offset);
    uint16_t readRegisterFromBuffer(int offset);
    void writeCoilToBuffer(int offset, int state);
    void writeRegisterToBuffer(int offset, uint16_t value);
    uint8_t writeStringToBuffer(int offset, uint8_t *str, uint8_t length);
    CallBackFunc cbVector[CB_MAX];
	void Modbus::setUnitID(uint8_t _unitID)
	{
		// set modbus slave unit id
		unitID = _unitID;
	}
private:
    Stream &serial;
    uint32_t timeout;
    uint32_t last_receive_time;
    uint16_t calcCRC(uint8_t *buf, int length);

    int ctrlPin = -1;
    uint8_t unitID;
    uint8_t lengthIn;
    uint8_t bufIn[MAX_BUFFER];
    uint8_t bufOut[MAX_BUFFER];
};
#endif
`

This is ok?

yaacov, 2018-12-31:
it's nicer to add the function in the `xxx.c` file, and only the header `void setUnitID(uint8_t _unitID);` in the `xxx.h` file

g6pl, 2018-12-31:
`
#define CTRL_PIN 4
#define BAUDRATE 9600
#define SLAVE_ID 5

Modbus slave(SLAVE_ID, CTRL_PIN);

void setup() 
{  
  SLAVE_ID = 12; //how i can change? This is no good.. :(
.
.
.
}
`

yaacov, 2018-12-31:
once you have the `setUnitID` method header defined in the `.h` and body defined in the `.c` file you can call it `slave.setUnitID(12)`

g6pl, 2018-12-31:
Thank you very much.
Everything works :)
Happy New Year!

yaacov, 2018-12-31:
> Happy New Year!

Happy New Year :tada:  :balloon:

closed 2018-12-31 by yaacov