yaacov/ArduinoModbusSlave #27: Won't compile for Teensy 3.x and LC when USB type is set to 'Serial' - Must have keyboard??

DrFrankReade, 2019-02-06:
Hello,

Getting bizzaro complier errors, probably a Teensy issue, but I never can tell because I'm not good at this stuff.

The simple.ino code won't compile on the Teensy 3.x and LC unless the "USB Type" includes a "keyboard"
Doesn't work under regular serial, doesn't work under NO USB. Only seems to work if Keyboard is in the mix. Here's the horrendous error:
```
Linking everything together...
"C:\\Program Files (x86)\\Arduino\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-gcc" -O2 -Wl,--gc-sections,--relax,--defsym=__rtc_localtime=1549453279 "-TC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\cores\\teensy3/mk20dx256.ld" -lstdc++ -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -o "C:\\Users\\andy\\AppData\\Local\\Temp\\arduino_build_341405/simple.ino.elf" "C:\\Users\\andy\\AppData\\Local\\Temp\\arduino_build_341405\\sketch\\simple.ino.cpp.o" "C:\\Users\\andy\\AppData\\Local\\Temp\\arduino_build_341405\\libraries\\ArduinoModbusSlave\\ModbusSlave.cpp.o" "C:\\Users\\andy\\AppData\\Local\\Temp\\arduino_build_341405/..\\arduino_cache_695210\\core\\core_teensy_avr_teensy31_usb_serial,speed_96,opt_o2std,keys_en-us_4939f5412a75a8560acb0f9e98268f02.a" "-LC:\\Users\\andy\\AppData\\Local\\Temp\\arduino_build_341405" -larm_cortexM4l_math -lm
c:/program files (x86)/arduino/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/lib/armv7e-m\libc.a(lib_a-writer.o): In function `_write_r':

writer.c:(.text._write_r+0x12): undefined reference to `_write'

collect2.exe: error: ld returned 1 exit status

Using library ArduinoModbusSlave at version 2.0.0 in folder: C:\Users\andy\OneDrive\Documents\Arduino\libraries\ArduinoModbusSlave 
Error compiling for board Teensy 3.2 / 3.1.

```

So it's all about this keyboard thing. 

I tried compiling in the Arduino IDE 1.8.8 / TeensyDuino 1.45

Also in Platform.io / Visual Code with the same results.

yaacov, 2019-02-07:
Hi, thanks for the issue !

a - I do not have a teensy board to play with :-)

b - from searching the web, did not try this myself ...  - 

b.1 - On teensy `Serial` is the usb port, you need to use one of the uart ports, e.g. `Serial1`.
https://www.pjrc.com/teensy/td_uart.html

b.2 - See examples here, for setting different serial ports:
```
// explicitly set stream to use the Serial serialport
Modbus slave(Serial, 1, 8); // stream = Serial, slave id = 1, rs485 control-pin = 8
```
https://github.com/yaacov/ArduinoModbusSlave#handle-read-input-registers-as-arduino-analogread

yaacov, 2019-09-03:
Closing because issue is too old ... please reopen when we get new data.

closed 2019-09-03 by yaacov