yaacov/ArduinoModbusSlave #35: Last release looks broken.

raf59, 2019-06-14:
Hi !
I tried the last release of the library and it looks broken.
I used the examples supplied with both ArduinoModbusSlave (full.ino) and node-modbus-serial (polling_RTU.js), and I wasn't able to make the communication work.
I only tried the ReadHoldingRegister function.

I used the following code :
```
#include "ModbusSlave.h"

#define SLAVE_ID 4
#define CTRL_PIN 8
#define BAUDRATE 9600

Modbus slave(SLAVE_ID, CTRL_PIN);
uint16_t value = 1234;

void setup() {
  pinMode(CTRL_PIN, OUTPUT);
  slave.cbVector[CB_READ_HOLDING_REGISTERS] = readMemory;
  Serial.begin( BAUDRATE );
  slave.begin( BAUDRATE );
  Serial.println("NEW_MODBUS_LIBRARY");
}

void loop() {
  if (slave.poll() != 0) {
    Serial.println(value);
  }
}

uint8_t readMemory(uint8_t fc, uint16_t address, uint16_t length) {
  slave.writeRegisterToBuffer(0, value);
  return STATUS_OK;
}
```

Here is a console record when starting the master :
[image]

As soon as I swap the new release with the previous one, everything works fine :
[image]

yaacov, 2019-06-14:
Thanks for the issue :+1:  @falahati  can you take a look ?

falahati, 2019-06-14:
Hey @raf59, thanks for the issue.
Please allow me to test it myself. I will report back.

falahati, 2019-06-15:
I just submitted a PR for this problem to the main repo. However, it would be great if you could try this before @yaacov merges it. The branch with this fix included is available here:

https://github.com/falahati/ArduinoModbusSlave/tree/fix2

yaacov, 2019-06-15:
merged :heart:

raf59, 2019-06-15:
Hello @falahati
Thanks for the PR. 😃  
I can't try it right now. My bus is busy, I'm performing another test wich takes a long time. 
I'll give it a try tonight

raf59, 2019-06-15:
I just tried the library at 9600 bauds.
When I just write `slave.poll()` in the loop, I only get CRC errors.

[image]

As soon as I write on the Serial line in the poll event (`slave.poll()!=0 { Serial.println(); }`), everything starts working.

[image]

I get the same behaviour up to 28800 bauds.
With higher baudrates, everything seems to be working fine, even without printing to the Serial line.

falahati, 2019-06-16:
Thanks for the detailed reports.
This seems like a transmission control problem. Probably `Serial` clears the transmission buffer before starting to send a byte.

If would be great if you check out this branch before I submit a PR:
https://github.com/falahati/ArduinoModbusSlave/tree/fix3

I also strongly suggest using another `Serial` port for RS485 communications instead of the board's main serial port (aka; the USB port). `AltSoftSerial` is the best alternative I found out based on the communication technics it uses if your board only provides you with one serial output. That being said, I am happy that you used the main serial port and therefore provided us with much-needed feedbacks.

PS:
Let me clarify what changed between the two releases.
In the older releases; `ArduinoModbusSlave` used to write the response in one go and then waited for it to get completely transmitted (if this functionality was available in the underlying `Serial` implementation); this is known as blocking communication. I changed this behaviour to nonblocking; meaning that in the new release; `ArduinoModbusSlave` now writes the response to the underlying `Serial` library's buffer in one or more steps and then periodically checks if the buffer is empty therefore transmission is considered as completed. 

There are clear advantages to this; one is the fact that it is interrupt friendly and also allows for larger transmissions as well as faster transmission due to the fact that majority of the `Serial` libraries block the write request if the requested data is larger than the transmission buffer of the `Serial` library at the time until the transmission buffer is completely empty. It is also watchdog friendly as it does not keep the program in a `while` loop as well as making your real code more responsive.

Now comes the cons; or more specifically; the con. Almost every `Serial` library has a bad nonblocking implementation. `AltSoftSerial` didn't even report the free space available in the transmission buffer up until the last week (still waiting for a release; the workaround is already a part of the code); `Serial` seems to empty the buffer before even completely sending the data (hence your current issue) and almost every one of them uses a different size for the buffer (workaround is, again, already a part of the code).

Still; I think is worth the time and resources to use the nonblocking technics if only for the big list of pros it offers.

yaacov, 2019-06-16:
@falahati  can you reproduce ? if yes :-) does this fix work on your side ?

falahati, 2019-06-16:
@yaacov; yes it does Yaacov.

yaacov, 2019-06-16:
> @yaacov; yes it does Yaacov.

That is good enough for me :sunglasses:  if you want to make a PR :cake:

closed 2019-09-03 by yaacov