yaacov/ArduinoModbusSlave #39: lack of communication from time to time

g6pl, 2019-09-03:
Hello
I have problem..
Every few minutes, no communication via rs485. How to reset the arduino is ok, again for about 5 minutes.
There is some command to reset only the library
Some splush? or reset port?

yaacov, 2019-09-03:
Hi, thanks for the issue,

I currently do not have h/w to check ... putting a help-wanted flag,
maybe someone having similar problem can help here.

p.s.
Adding  @falahati @TobiasSchaffner because they where the last people I know that worked on this, maybe they know something.

yaacov, 2019-09-03:
@g6pl can you attach / link a wiring description of your rs485 module, it may be fixable by changing something in hardware design.

g6pl, 2019-09-03:
I noticed that the communication freezes when the query from the master is long, or about a large amount of data from the slave...

yaacov, 2019-09-03:
does changing buffer size, do anything bad/good ?

https://github.com/yaacov/ArduinoModbusSlave/blob/master/src/ModbusSlave.h#L22

g6pl, 2019-09-03:
Can you limit the length of the query? So that it does not respond to more than, e.g. 32 characters?

yaacov, 2019-09-03:
you can try return some error status if length is too big ...

falahati, 2019-09-03:
Hey @g6pl,
Can you please provide us with more information regarding the request that is too long? Like the function id, offset and count.

g6pl, 2019-09-04:
I noticed that there is an array overflow (array_writeMemory) and then it happens
My array is: 
float array_writeMemory[40];
My code:
`uint8_t writeMemory(uint8_t fc, uint16_t address, uint16_t length) {
    
   
    uint16_t registerIndex;
    // write to eeprom.
    for (int i = 0; i < length; i++) 
    {
        array_writeMemory[i + address]=slave.readRegisterFromBuffer(i);
    }

          
        return STATUS_OK;
}`

yaacov, 2019-09-04:
so if length is larger then 40, you can return some error status, e.g. `STATUS_ILLEGAL_DATA_ADDRESS` , 
does that help you case ?

g6pl, 2019-09-04:
`uint8_t writeMemory(uint8_t fc, uint16_t address, uint16_t length) {

uint16_t registerIndex;
// write to eeprom.
for (int i = 0; i < length; i++) 
{
    if (length < 40){tablica_writeMemory[i + address]=slave.readRegisterFromBuffer(i);}
        else{return STATUS_ILLEGAL_DATA_ADDRESS; }
}

      
    return STATUS_OK;
}`

Nothing helps :(
If i send " 01  10  00 00  00 1e"
transmission is freezing

yaacov, 2019-09-04:
should be `(i + address) < 40` ?

g6pl, 2019-09-04:
`if (((i + address) < 40)){array_writeMemory[i + address]=slave.readRegisterFromBuffer(i);}
        else{return STATUS_ILLEGAL_DATA_ADDRESS; }`

Nothing helps :(
I send: 01  10  00 00  00 0a - is ok
I send: 01  10  00 00  00 0b - is ok
i send: 01  10  00 00  00 0c - NOT OK and transmission is freezing

g6pl, 2019-09-04:
Hey, i use your old library....
In new library is ok.......

yaacov, 2019-09-04:
Greate :cake:

closed 2019-09-04 by yaacov