yaacov/ArduinoModbusSlave #45: Is it possible to change parity in 485 modbus comunication?

jagonga, 2020-01-30:
Dear Mr. Yaacov,
I would like to know if is it possible to change parity in comunication.
For example, I can connect with 9600-8-N-1, but if I want to connect with 9600-8-O-1 (odd parity), how I can do it?
I look forward to hearing from you soon.
Thanks.

Kind regards,
Javier

yaacov, 2020-01-30:
Hi, thanks for the issue.

The modbus connection can use the arduino serial port.

Setting and opening the port in the example:
https://github.com/yaacov/ArduinoModbusSlave/blob/master/examples/simple/simple.ino#L56

Setting the parity in arduino documentation:
https://www.arduino.cc/reference/en/language/functions/communication/serial/begin/

So to set the serial connection to`9600-8-O-1`
```
Modbus slave(SLAVE_ID, CTRL_PIN);

void setup() {
    Serial.begin( 9600, SERIAL_8O1 );
    slave.begin( 9600 );
}

```

jagonga, 2020-01-30:
Dear Yaacov,
You are right, the solution is very simple.
Thanks.

Kind regards,
Javier

closed 2020-01-30 by yaacov