yaacov/ArduinoModbusSlave #46: win10 Codesys ControlWin softplc

zatalian, 2020-02-12:
Hi,

I have a codesys controlwin softplc configured as modbus master. Communication only works if i use a arduino leonardo as slave hardware. The exact same project on a arduino uno or arduino duemilanove does not work.
Is this a know problem (and maybe a solution/workaround)?

yaacov, 2020-02-12:
Hi, thanks for the issue :+1:

> Is this a know problem (and maybe a solution/workaround)?

a. no, if you fix it, please send a pull request fixing it.
b. if you post the relevant code that fails on uno and works on leonardo here, maybe someone can see the problem and help fix the issue.

zatalian, 2020-02-12:
the default examples all work with the combination codesys + leonardo.
I suspect the different usb to serial drivers used by the different arduino's but i haven't found a solution yet.

ProAce, 2020-02-12:
Quick sanity check:
1. Did you change the Serial port to Serial1 for the leonardo and Serial for the uno?
2. Did you disconnect the USB on the uno when testing? The Serial port is shared between the two

zatalian, 2020-02-12:
I don't use an RS485 shield. I communicate directly over the usb comport. So I use Serial on both the leonardo and the uno.

yaacov, 2020-02-25:
@zatalian any news ?

zatalian, 2020-04-15:
no solution yet but i don't think it is the arduino code that is causing the problem. I have similar problems with another program communicating with a leonardo and an uno. For the uno, i have to disable RTS and DTR. Something i can't do with codesys...

yaacov, 2020-04-15:
:+1: thanks