yaacov/ArduinoModbusSlave #49: ModbusTCP

ProAce, 2020-03-31:
For a project I would like to use ModbusTCP. For this I looked into using your TCP lib, but this only accepts a ESP8266 and is about 3 years old. Would it be worth the effort to fix that library, or would it be better to change this one to also accept TCP messages?

Would like to know your vision on this problem.

yaacov, 2020-03-31:
Hi, thanks for the question.

> Would it be worth the effort to fix that library, or would it be better to change this one to also accept TCP messages?

I think the easier approach will be to make the esp8266 version work for you hardware.
I think the better solution ( will require more work ) is to add pluggable h/w options to this lib, starting with whatever h/w you have :-)

ProAce, 2020-03-31:
I would like to make this work for all situations (as far as possible). So I will try and change this lib. If I get this working, would that be a feature you would want to add to the library?

yaacov, 2020-03-31:
> would that be a feature you would want to add to the library?

Sure :+1:

yaacov, 2020-03-31:
p.s.

> So I will try and change this lib

Try to make it easy to add support for other h/w with you hardware as first example, so you will not need to change current code.

ProAce, 2020-03-31:
My idea is to incorporate a similar method as used with this rewrite: https://github.com/AdvancedClimateSystems/sps30

I'm just not yet sure if the different server classes can be used just like a Stream object.

yaacov, 2020-03-31:
have fun :green_heart:

falahati, 2020-05-14:
@ProAce,
If you decided to go with this library for Modbus TCP make sure to keep the memory aspect of the program in mind. Parsing TCP can easily get out of hand especially when the library is created to handle both serial and TCP. Personally I suggest a separate fork for TCP if you decided to do so.

ProAce, 2020-05-14:
I did start working on it, but currently I don't have much time or need for it to finish it. 

The memory aspect was one thing I ran into, but it seems possible to reuse the space for the RTU messages for TCP messages as the only difference are the headers. Managing this however in a neat fashion asked for a more structured data tagging which needed a bit more rewriting than I anticipated.