yaacov/ArduinoModbusSlave #57: Holding Registers

DpunktS, 2020-04-29:
Hello I have described an Arduino pro mini ATmega328p with the full.ino in Arduino IDE. Now my problem I can not choose between digital inputs and outputs, the holding register always shows 255 when reading but writing does not always result in error: timeout with QModMaster and the digital pins stay on output.

ProAce, 2020-05-01:
When your EEPROM is empty before uploading the full.ino example all the digital pins should be INPUTS. This means that if you read the Input Status this should reflect the current state of the digital pins you defined. To change the pinMode you should write a 0 (input) or a 1 (output) to the corresponding Holding Register.

From your description I'm not completely sure what it is that you're trying to do.

DpunktS, 2020-05-01:
At the very beginning, all PINs were inputs.  Then I wanted to write in the holding register.  Writing always stops with an error ubd when reading the honding register the next time there were 255 in it and all pins sibd outputs.  how do I get the 255 out of the eeprom.

yaacov, 2020-05-03:
This is the loop that read the eeprom
https://github.com/yaacov/ArduinoModbusSlave/blob/master/examples/full/full.ino#L52

This is the line handling `CB_WRITE_HOLDING_REGISTERS` that changes the `pinMode`
https://github.com/yaacov/ArduinoModbusSlave/blob/master/examples/full/full.ino#L210

so you can:
a. set the eeprom to what needed before the loop in line 52
b. call the device with modbus `write holding register` to the address that will change the currect binMode.

DpunktS, 2020-05-03:
I have now deleted the EEPROM again so that it is at 0. Nevertheless I cannot write 1 into the EEPROM with QModMaster. Error code: Illegal data value

ProAce, 2020-05-03:
https://github.com/yaacov/ArduinoModbusSlave/blob/4fbc8c155a5752e532990661089922e167586042/examples/full/full.ino#L201 This line should have a `&&` instead of a `||`. Small logic mistake that breaks this example. I'll make a merge request to fix this.

closed 2020-05-03 by yaacov