yaacov/ArduinoModbusSlave #60: Doesn't work on arduino mega 2560.

gyuseokByeon, 2020-05-10:
Modbus :: createResponse (); If there is no delay (1) or Serial.println ("") before calling the function, it does not work properly.

uint8_t Modbus::poll()
{
.....
    // Validate the incoming request.
    if (!Modbus::validateRequest())
    {
    	  Serial.println("notvalid");
        return 0;
    }
   //Serial.println("valid");
    //delay(1);
    // Execute the incoming request and create the response.
    uint8_t status = Modbus::createResponse();
}

falahati, 2020-05-14:
Please provide information about what exactly happens. Does the library read the full request and the `validateRequest()` method passes? Does your callback gets executed? Do you get bad response or no response? What is your callback function, please post it here.

GopalN, 2020-09-24:
This bug i also can verify.I am running the example code named SImple.ino which comes as part of the package.
When i did the above modification it started working else the code used to crash and reset itself.But if i add more that 2 Callbacks it again stops working but this time does not crash.

    slave.cbVector[CB_WRITE_COILS] = writeDigitalOut;
    //slave.cbVector[CB_READ_DISCRETE_INPUTS] = readDigitalIn;//Enabling this makes the code stop working
    slave.cbVector[CB_READ_INPUT_REGISTERS] = readAnalogIn;
    //slave.cbVector[CB_READ_HOLDING_REGISTERS]=writeAnalogOut//Enabling this makes the code stop working;

`#include <ModbusSlave.h>

#define SLAVE_ID 1           // The Modbus slave ID, change to the ID you want to use.
#define SERIAL_BAUDRATE 9600 // Change to the baudrate you want to use for Modbus communication.
#define SERIAL_PORT Serial1   // Serial port to use for RS485 communication, change to the port you're using.

// Comment out the following line if your not using RS485
#define RS485_CTRL_PIN 8 // Change to the pin the RE/DE pin of the RS485 controller is connected to.

// The position in the array determines the address. Position 0 will correspond to Coil, Discrete input or Input register 0.
uint8_t input_pins[] = {7, 9, 10};     // Add the pins you want to read as a Discrete input.
uint8_t output_pins[] = {3, 5, 6};   // Add the pins you want to control via a Coil.
uint8_t analog_pins[] = {A0, A1, A2}; // Add the pins you want to read as a Input register.

// You shouldn't have to change anything below this to get this example to work

uint8_t input_pins_size = sizeof(input_pins) / sizeof(input_pins[0]);    // Get the size of the input_pins array
uint8_t output_pins_size = sizeof(output_pins) / sizeof(output_pins[0]); // Get the size of the output_pins array
uint8_t analog_pins_size = sizeof(analog_pins) / sizeof(analog_pins[0]); // Get the size of the analog_pins array

#ifdef RS485_CTRL_PIN
// Modbus object declaration
Modbus slave(SERIAL_PORT, SLAVE_ID, RS485_CTRL_PIN);
#else
Modbus slave(SERIAL_PORT, SLAVE_ID);
#endif

unsigned long Alive_poll=0;
void setup()
{
    Serial.begin(115200);
    // Set the defined input pins to input mode.
    for (int i = 0; i < input_pins_size; i++)
    {
        pinMode(input_pins[i], INPUT);
    }

    // Set the defined analog pins to input mode.
    for (int i = 0; i < analog_pins_size; i++)
    {
        pinMode(analog_pins[i], INPUT);
    }

    // Set the defined output pins to output mode.
    for (int i = 0; i < output_pins_size; i++)
    {
        pinMode(output_pins[i], OUTPUT);
    }

    // Register functions to call when a certain function code is received.
    slave.cbVector[CB_WRITE_COILS] = writeDigitalOut;
    //slave.cbVector[CB_READ_DISCRETE_INPUTS] = readDigitalIn;
    slave.cbVector[CB_READ_INPUT_REGISTERS] = readAnalogIn;
    //slave.cbVector[CB_READ_HOLDING_REGISTERS]=writeAnalogOut;
    // Set the serial port and slave to the given baudrate.
    SERIAL_PORT.begin(SERIAL_BAUDRATE);
    slave.begin(SERIAL_BAUDRATE);
    Serial.println("MODBUS SLAVE STARTED AT SLAVE1");
    Alive_poll=millis();
}

void loop()
[... 68 lines cut ...]
    {
        // Write the state of the analog pin to the response buffer.
        slave.writeRegisterToBuffer(i, analogRead(analog_pins[address + i]));
    }

    return STATUS_OK;
}

// Handle the function code Read Input Registers (FC=04) and write back the values from analog input pins (input registers).
uint8_t writeAnalogOut(uint8_t fc, uint16_t address, uint16_t length)
{
    // Check if the requested addresses exist in the array
    if (address > output_pins_size || (address + length) > output_pins_size)
    {
        return STATUS_ILLEGAL_DATA_ADDRESS;
    }

    // Read the analog inputs
    for (int i = 0; i < length; i++)
    {
        analogWrite(output_pins[address + i], slave.readRegisterFromBuffer(i));
    }

    return STATUS_OK;
}`