yaacov/ArduinoModbusSlave #65: Is it possible to use in ESP32 Platform.

coderkan, 2020-08-28:
Hi there,

I wonder that can I use this library on ESP32 module? 
Have you ever tried that?

Regards.
Erkan

yaacov, 2020-08-30:
Hi, thanks for the question/

I don't know, if you make it work comment here.
Matve this repo is better starting point:
https://github.com/yaacov/ArduinoModbusSlaveTCP

coderkan, 2020-09-08:
Hi @yaacov ,

I tried to use in ESP32 kit but I have no success at this point. I focused on Serial RTU.
I will try again later. If It would work, I will comment you.

Regards.
Erkan

Bach-Danielsson, 2020-09-24:
Ler me know if someone have sucsess with this. im also srugling to get the RTU slave to Work on my ESP32

coderkan, 2020-09-26:
Hi @Bach-Danielsson 

I have not success to run this code in ESP32.

I have found a [library](https://github.com/bertmelis/esp32ModbusRTU)  on github. But I have come across problems to use this library.
Please let me know if you find a good solution for rtu slave for esp32.

Regards.
Erkan

device111, 2021-09-13:
Hi,

I've tested the RTU-RS485 Client and with a few tweaks the code works like a charm with the ESP32dev Kit.
I make a Pull Request after complete testing. (need a few days...)

Martin

547176052, 2022-04-12:
esp32能用,但是0x03每次最多只能读50个,0x0f每次最多只能写50个

yoonghm, 2022-04-13:
This is limited by Modbus protocol.

device111, 2022-04-13:
that is not completely right. The Modbus protocol has a frame width of 256 bytes. (253 bytes for user data). We are currently only using 128 bytes here, according to the specifications of the ESP32 core. You can try changing the RX/TX buffer size.

In ModbusSlave.h change:

`#if defined (ESP32) || defined (ESP8266)
   #define SERIAL_BUFFER_SIZE 256
#endif`

and in your Setup Routine and **BEVORE** the Serial init:

`   SERIAL_PORT.setRxBufferSize(SERIAL_BUFFER_SIZE);`
`    SERIAL_PORT.setTxBufferSize(SERIAL_BUFFER_SIZE);` 

`    // Set the serial port and slave to the given baudrate.`
`    SERIAL_PORT.begin(SERIAL_BAUDRATE);`
   ` slave.begin(SERIAL_BAUDRATE);`

I don't know if that works with the Arduino 1.6 ESP Core, but it should work with the 2.02.

good luck...

547176052, 2022-04-17:
#if defined (ESP32) || defined (ESP8266)
  #define SERIAL_BUFFER_SIZE 1024
#endif

我改到1024还是只能写 55 个

device111, 2022-04-18:
No Problem to read 125 Words of an 03 Register:

[image]

so your Master is not support more Register?

dtboy1995, 2022-07-03:
> No Problem to read 125 Words of an 03 Register:
> 
> [image]
> 
> so your Master is not support more Register?

Can you tell me what software this is, thank you

device111, 2022-07-04:
[image]

dtboy1995, 2022-07-04:
> [image]

thanks

tinkering4fun, 2023-02-05:
I'm successfully using V2.1.1 on ESP32 and it worked more or less out of the box ...
Only minor adjustments required to get it running. See issue #98.