yaacov/ArduinoModbusSlave #68: Missing support for arduino nano every

salvitti, 2020-10-08:
There are problems to compile the library for the new Arduino Nano Every target.
They declared that Arduino Nano Every is compatible with Arduino Nano but that's not true.
It seems the Arduino Nano Every doesn't suppport software serial but only hardware serial.
Could it possible to modifiy the library for supporting the Arduino Nano Every too?

yaacov, 2020-10-08:
Hi, thank you for the issue.

I don't have a Arduino Nano Every to test, if you make the library work with your hardware, please make a pull request with the fix.

Added a "help wanted" label in case someone that has the h/w or can guess a fix, can help.

falahati, 2020-10-08:
This library works with h/w serial and since Nano Every does seem to have two separate h\w serial ports, you should be okay with using the USB port at the same time.

try

```C
Modbus slave(Serial1, 1, 8); // slave id = 1, rs485 control-pin = 8
```
or
```C
Modbus slave(SERIAL_PORT_HARDWARE , 1, 8); // slave id = 1, rs485 control-pin = 8
```

ProAce, 2020-10-08:
The library wouldn't compile because the max buffer length define was different. Small change to the architecture check fixes it