yaacov/ArduinoModbusSlave #74: Stuck when calling flush() in begin() when using Serial1 in Arduino MKR(Zero)

patriksmedberg, 2021-01-11:
Using an Arduino MKRZero I get stuck in row 190 calling flush() when using Serial1. If I send dummy byte on the Serial1 before calling .begin() it works.

Seems the blame is in Arduino Core. 
https://github.com/arduino/ArduinoCore-samd/blob/ad95d3c204d43bee789bb0c116afbec135e6ace5/cores/arduino/SERCOM.cpp
void SERCOM::flushUART()
{
  // Skip checking transmission completion if data register is empty
//   if(isDataRegisterEmptyUART())
//     return;

  // Wait for transmission to complete
  while(!sercom->USART.INTFLAG.bit.TXC);
}

yaacov, 2021-01-11:
Thank you for the issue :green_heart: 

can you add a pull request that will fix that ?

yaacov, 2021-04-21:
fixed by #77

closed 2021-04-21 by yaacov