yaacov/ArduinoModbusSlave #75: Not working with I2C communication

akshulg, 2021-03-01:
I had successfully implemented this library with Arduino as an I2c slave. So one arduino modbus slave + I2c slave works. But right now I have interfaced a I2C sensor with arduino, and modbus is not working, not able to scan registry on a modbus master end. Arduino modbus slave + I2C master won't work. I have tried with some flags but still not working. Any ideas on what might be the issue?

yaacov, 2021-03-01:
Hi, thank you for the issue.

Both I2C and Modbus are buses that solve the problem of master/slaves communication, so when using the same line you will have two masters, one for I2C and one for Modbus, My guess guess is that they want to talk/recive the same time and interfere.

akshulg, 2021-03-01:
@yaacov yes that's what I observe, that both are trying to communicate and hence there is conflict. Is there any way to implement both things? 

I am trying external interrupt right now. So I have connected the INT1 (pin 3) of arduino to RX of arduino. RX pin is connected to TTL to RS485 converter pin. So when arduino receive the request from a modbus master then interrupt should trigger and sensor (which is on I2C) data should poll. But what I observe is that the RX pin is always on. Any idea on this method?

yaacov, 2021-03-01:
> Is there any way to implement both things?

I don't think so, even if you can shut down one server, it's clients will continue to interfere with the other bus communication

akshulg, 2021-03-01:
Do you have any other idea that I can try and implement?

falahati, 2021-03-01:
use a random serial library for the Modbus if you can afford another line. (e.g. AltSoftSerial)

akshulg, 2021-03-01:
I am already using AltSoftSerial for debugging and programming purpose. Programming of slave id, baud rate and other parameters. But can interchange Serial and AtlSoftSerial purpose. Will try that.

akshulg, 2021-03-02:
It's not working with AltSoftSerial as well. I am stuck now with no ideas!

akshulg, 2021-03-02:
@falahati  It's not working with AltSoftSerial as well. I am stuck now with no ideas!

Apulanta, 2021-03-02:
I have used a working solution Arduino Nano Modbus TX (0), 1 / RX (1), and
the I2C bus A4 (13) of the SDA, A5, (14), SCL

akshulg, 2021-03-03:
@Apulanta That's the same configuration I am using with Arduino UNO. But it's not working.