yaacov/ArduinoModbusSlave #76: Enable / disable communication

cvanbrederode, 2021-04-20:
I have a very specific feature request for a product our company has produced.  I implemented it (and changing the Unit ID) easily enough, but thought I would add an issue.  I can issue a PR with my version of it if needed.

The ability to turn communication on and off with a function call.  Our product doesn't communicate at start up until a digital in is raised, then it talks on ID 1.  When the digital in is lowered, it switches to a different ID according to a register.  This allows us daisy chain multiple units together without having to pre-configure them; the master gives them each an address in turn.

My fix was a simple private bool, and some methods to toggle it.  If the bool is false, processing of packets would stop before checking the unit Id.

yaacov, 2021-04-20:
hi, thank you for the issue

a pull request with the feature will be very welcome.

cvanbrederode, 2021-04-21:
Pull request created.  Had to get back into work to test the changes.