yaacov/ArduinoModbusSlave #81: using a modbus master simulator software with this library

mushfiq-heraclius-xen, 2021-07-12:
can any body tell me which modbus simulator software is better to use with this library , i have used "Simply Modbus Master " simulator and tried writing/reading data to register number 40001 , didn't get any response or anything

ProAce, 2021-07-12:
What kind of program where you running for this test?

mushfiq-heraclius-xen, 2021-07-12:
https://www.simplymodbus.ca/RTUmaster.htm

ProAce, 2021-07-12:
I meant the program on the microcontroller, sorry for the confusion.

mushfiq-heraclius-xen, 2021-07-12:
https://github.com/yaacov/ArduinoModbusSlave/tree/master/examples/simple

ProAce, 2021-07-12:
The way that example is set up it isn't possible to read anything at register 40001. You'd have better luck when reading registers 0,1,2 with fc 5. This reads out the set analog input values.

This library can adhere to the standard offsets, but doesn't do so in the examples. This ability to use any registry makes it quite flexible.

mushfiq-heraclius-xen, 2021-07-12:
ok thank you! but can you tell me what values i have to write for fc=05 or 15 to make output pins HIGH ??

mushfiq-heraclius-xen, 2021-07-12:
Force Single Coil (FC=05) and Force Multiple Coils (FC=15) and set the corresponding digital output pins (coils).

mushfiq-heraclius-xen, 2021-07-12:
what values i have to write from the modbus rtu master simulator software for fc=05 or 15 to make output pins HIGH ??

ProAce, 2021-07-12:
I was mistaken, for reading the analog sensors FC04 should be used.

When writing a 1 to register 0 to 3 when using FC05 the digital outputs should be set to high.

paxsipornax, 2021-11-21:
readRegister = slave.readCoilFromBuffer(40001) How can I write a code like
I want to read the data in plc 40001 register but failed