yaacov/ArduinoModbusSlave #95: Create new release

maxgerhardt, 2022-10-06:
The current release (https://github.com/yaacov/ArduinoModbusSlave/releases) is 2.0.0 from 2017. A lot of things have changed since then. People downloading this library via the Arduino IDE library manager or PlatformIO (https://registry.platformio.org/libraries/yaacov/ModbusSlave/versions) get the latest stable as that 2017 version, with which the examples shown in the Github of course not work.

Can you create a new stable release, e.g. 2.1.0, so that people can get the latest version as a regular release version?

maxgerhardt, 2022-10-18:
The version in `library.properties` was not updated, so all library crawlers will not see this new version:

https://github.com/yaacov/ArduinoModbusSlave/blob/v2.1.0/library.properties#L2

yaacov, 2022-10-18:
2nt try :-)
https://github.com/yaacov/ArduinoModbusSlave/releases/tag/v2.1.1

maxgerhardt, 2022-10-19:
Works! https://registry.platformio.org/libraries/yaacov/ModbusSlave/versions

[image]

closed 2022-10-19 by maxgerhardt