yaacov/ArduinoModbusSlave #98: ESP32 Support

tinkering4fun, 2023-02-03:
Hi Yaacov,
thanks for this great Modbus Slave library (the best I found so far for building a Modbus RTU Slave).

While using the code on **ESP32** I found a minor regression from issue #97:
The signature of the callback functions has changed, while the examples do not yet adopt to it.
Easy to fix, just append the missing ```, void *context``` in the signature of the example's callback functions.
(Btw. When compiling for AVR there is no error on compile time, but I didn't try to run the code on AVR.)

Besides this, I would suggest another small fix in ModbusSlave.cpp:
The class constructor should initialize all ```cbVector[]``` entries with NULL.
To be honest, I'm currently unsure about the best single (?) place in the constructor(s) where to do it. 
Without the init the slave crashes when a Modbus request is received, with a FC where no appropriate callback vector has been set for. 
Because the  ```if(callback)``` in the code snippet below does not evaluate to false for this case, maybe this is also just a ESP32 issue.
```
860: uint8_t Modbus::executeCallback() {
...
865: ModbusCallback callback = _slaves[i].cbVector[callbackIndex];
...
868: if (callback)
...
875: if (callback)
```

What do you think about official  **ESP32 support** for your code?
The benefit with ESP32 as Modbus RTU Slave is imho the 2nd hardware Serial port available (and of course some projects may also benefit from WiFi etc.).
See my working ESP32 Slave example code below (derived from simple.ino).
Requires the trivial fix in ModbusSlave.cpp desribed above.

Kind regards
Werner

[simpleESP32.ino.txt](https://github.com/yaacov/ArduinoModbusSlave/files/10582720/simpleESP32.ino.txt)

yaacov, 2023-02-05:
Thank you for the issue

tinkering4fun, 2023-02-18:
Update: I try to re-sort my fork into different branches ...

- issue98: is the fix where this issue is about.

The rest goes way beyond this ...

- kernel: A class which already provides some which I would expect from any of my workable slaves.
The idea is to have a consistent set of kernel functionality and code the dedicated application functions in a derived class.
- kernel-examples: Examples using the above concept.
- allnew: Merging all of the above

- EP32: Will finally only contain the addition in library.properties file.  ESP32 architecture is currently not listed, but the library seems to work.

Erfan12699, 2023-06-20:
thanks dear Werner

I want to put an ESP32 module as a slave by Yaacov's library (Modbusslave.h) and read the analog data by Modbus pull in my pc.

can you help me?

Erfan12699, 2023-06-20:
dear Werner.
Does this library (Modbusslave.h) work on STM32 (bule pill)?

tinkering4fun, 2023-06-21:
Hi Erfan,
i guess my mentioned example  [ArduinoModbusSlave-ESP32-fork](https://github.com/tinkering4fun/ArduinoModbusSlave-ESP32-fork/tree/specific-examples)/[examples](https://github.com/tinkering4fun/ArduinoModbusSlave-ESP32-fork/tree/specific-examples/examples)/[specific](https://github.com/tinkering4fun/ArduinoModbusSlave-ESP32-fork/tree/specific-examples/examples/specific)/[ESP32](https://github.com/tinkering4fun/ArduinoModbusSlave-ESP32-fork/tree/specific-examples/examples/specific/ESP32)/[plainSlave](https://github.com/tinkering4fun/ArduinoModbusSlave-ESP32-fork/tree/specific-examples/examples/specific/ESP32/plainSlave)
does exactly what you asked for ...?

See line 445
` slave->writeRegisterToBuffer(i, analogRead(adc_pins[address + i]));` 

I assume you are already familiar with Modbus and have some Modbus client software at your PC-
If so,  just use FC04 to read the input registers 0 and 1, which will return the values from pins  ADC_0 and ADC_1.

There is also a straight-forward and simple example in Yaacov's lib
[ArduinoModbusSlave](https://github.com/yaacov/ArduinoModbusSlave/tree/master)/[examples](https://github.com/yaacov/ArduinoModbusSlave/tree/master/examples)/[simple](https://github.com/yaacov/ArduinoModbusSlave/tree/master/examples/simple)

It is made for Arduino board, but the code should also run on ESP32 after changing the ADC pin definitions.  
Just 150 lines of code, so it should be a good starter.

I never used STM32, if you own one, just give it a try and see  ...

btw:
If you never used Modbus before, it may be too complex to learn for just just pulling ADC values into an PC.
Sorry, but I will not guide you to a final Modbus end to end solution for your specific problem.

Hope this helped to some extend.

Regards
Werner

Erfan12699, 2023-06-22:
Thanks dear Werner.

yes, i used Yaacov's lib and read analog on Modbus pull.

i could use from Yaakov's library on ESP32 with your guide. very good issue.

thank you and dear Dr zemir

Regards
Erfan